anmaped/esp8266-scheduler #1: compile error

gaerank, 2017-04-17:
Hi, 
First of all, Thank you for sharing your code.
I have an experience error while compile with your sample code.
Could you help me out to figure this?

Thank you.

IDE : Arduino IDE 1.8.1

Error Messages :
arduino.ar(core_esp8266_wiring.c.o): In function `delay':
/home/kevin/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring.c:42: multiple definition of `delay'
libraries/esp8266-scheduler-master/Scheduler.cpp.o:/home/kevin/Arduino/libraries/esp8266-scheduler-master/Scheduler.cpp:97: first defined here
arduino.ar(core_esp8266_main.cpp.o): In function `esp_yield':
/home/kevin/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp:82: multiple definition of `esp_yield'
libraries/esp8266-scheduler-master/Scheduler.cpp.o:/home/kevin/Arduino/libraries/esp8266-scheduler-master/Scheduler.cpp:90: first defined here
arduino.ar(core_esp8266_main.cpp.o): In function `optimistic_yield':
/home/kevin/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp:104: multiple definition of `optimistic_yield'
libraries/esp8266-scheduler-master/Scheduler.cpp.o:/home/kevin/Arduino/libraries/esp8266-scheduler-master/Scheduler.cpp:95: first defined here
libraries/esp8266-scheduler-master/Scheduler.cpp.o:(.text._ZN14SchedulerClass5yieldEv+0x4): undefined reference to `setjmp'
libraries/esp8266-scheduler-master/Scheduler.cpp.o:(.text._ZN14SchedulerClass5yieldEv+0x8): undefined reference to `longjmp'
libraries/esp8266-scheduler-master/Scheduler.cpp.o: In function `SchedulerClass::yield()':
Scheduler.cpp:(.text._ZN14SchedulerClass5yieldEv+0x18): undefined reference to `setjmp'
Scheduler.cpp:(.text._ZN14SchedulerClass5yieldEv+0x2e): undefined reference to `longjmp'
libraries/esp8266-scheduler-master/Scheduler.cpp.o: In function `SchedulerClass::init(void (*)(), void (*)(), unsigned char const*)':
Scheduler.cpp:(.text._ZN14SchedulerClass4initEPFvvES1_PKh+0x1f): undefined reference to `setjmp'
collect2: error: ld returned 1 exit status
exit status 1
board ESPino (ESP-12 Module) compile error.

...
exit status 1
board Generic ESP8266 Module compile error.

...
exit status 1
board NodeMCU 1.0 (ESP-12E Module) compile error.

anmaped, 2017-04-28:
Hi @gaerank,

Yes. You need to apply a path on esp8266 runtime. I will provide it asap. I want to make this changes on arduino esp8266. But I will provide a path.

tymons, 2017-06-09:
Same here, what do you mean "to apply a path on esp8266 runtime"

Thanks!

Raphyyy, 2017-06-12:
Same here ! Some explanation would be welcome

anmaped, 2017-06-12:
@tymons @Raphyyy There are some files in the runtime of the arduino esp8266 that need to be replaced with some changes. What version of arduino esp8266 are you using?

tymons, 2017-06-12:
@anmaped I am using NodeMCU 1.0 (ESP12-E module) with Arduino Studio 1.6.12.

Raphyyy, 2017-06-12:
ESP12E with the latest arduino core on Platformio here

anmaped, 2017-06-13:
@tymons @Raphyyy I will update for the current arduino core version in two or three days.

kevinkk525, 2017-07-11:
Any progress for IDE 1.8.2? Having the same error.

Raphyyy, 2017-08-28:
Yep, still got the same error and didn't find any solution...

Kaisha, 2017-08-28:
Yup same error here.  Using the Adafruit Huzzah board.  I tried modifying platform.txt to add -Wl,-z,muldefs because clearly your're replacing the normal functions with your own versions, but that still left the undefined setjmp/longjmp.  As this point I'm just not too sure of the whole Arduino/ESP8266 toolchain (I do most of my coding with MSVC toolchain) to get any farther.  Even if you don't have time to make the changes, just a simple description of what needs to be done and I think I and many others can figure it out.

anmaped, 2017-10-06:
Core diff for esp8266 arduino revision 4897e0006b5b0123a2fa31f67b14a3fff65ce561. Use only that version of the esp8266 core. Patch for the new version soon.

Several changes on spiffs and umm_malloc.

Commit https://github.com/anmaped/esp8266-scheduler/commit/498112d0ca3d16b85838c1ab9a2a34174a52155c

srmq, 2018-01-04:
Hello, I also had issues to apply the patches, finally I was able to apply it. I made a branch to the esp8266 core with the patch applied here: https://github.com/srmq/Arduino/tree/scheduler_branch

anmaped, 2018-01-04:
@srmq  Thanks. I want to update them on the current Arduino version. If you are able to try that then please tell me something.

shiftleftplusone, 2018-02-07:
Same to me, not even SchedulerBlink can be compiled for esp8266.
I used the latest esp8266 cores from Arduino Board manager and the IDE 1.8.5, board nodeMCU 1.0
So is there a file which just has to be c+p'ed to where ever to make that thing run?
What is it about the file 
core_esp8266_changes
in the zip file?
How to open or where to copy?

anmaped, 2018-02-07:
@shiftleftplusone Please use this branch of Arduino esp8266 (https://github.com/srmq/Arduino/tree/scheduler_branch) with the patch files applied. @srmq  has done it for us!

shiftleftplusone, 2018-02-07:
sorry, I have no idea what to do with this link. They even refer to Arduino 1.6.4, but I have 1.8.5, and I also don't use git close or Python.

shiftleftplusone, 2018-02-13:
still does not work.
folder  esp8266-scheduler-master renamed to esp8266-scheduler and copied into the potable\libraries folder
Sketch Scheduler.blink copied to d:\ for current use.

Code: Scheduler.blink
```
/**
 * @file SchedulerBlink.ino
 * @version 1.0
 *
 * @section License
 * Copyright (C) 2015-2016, Mikael Patel
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * @section Description
 * This Arduino sketch uses the Scheduler library to blink four
 * LEDs with different speed. Each LED is controlled by a separate
 * task.
 */

#include <Scheduler.h>

const int led = 13;

void setup()
{
  pinMode(led, OUTPUT);

  Scheduler.start(setup1, loop1);
  Scheduler.start(setup2, loop2);
  Scheduler.start(setup3, loop3);
}

void loop()
{
  blink(led, 1000);
}

const int led1 = 10;

void setup1()
{
  pinMode(led1, OUTPUT);
}

void loop1()
{
  blink(led1, 250);
}

const int led2 = 11;

void setup2()
{
  pinMode(led2, OUTPUT);
}

void loop2()
{
  blink(led2, 500);
}
[... 16 lines cut ...]
  delay(ms);
  digitalWrite(pin, LOW);
  delay(ms);
}
```
errors:
Arduino: 1.8.5 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Program Files (x86)\Arduino\portable\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Program Files (x86)\Arduino\portable\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\portable\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path C:\Users\hannah\AppData\Local\Temp\arduino_build_116793 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\esptool\0.4.9 -verbose D:\esp8266-scheduler-master\esp8266-scheduler\examples\SchedulerBlink\SchedulerBlink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Program Files (x86)\Arduino\portable\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Program Files (x86)\Arduino\portable\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\portable\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path C:\Users\hannah\AppData\Local\Temp\arduino_build_116793 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\esptool\0.4.9 -verbose D:\esp8266-scheduler-master\esp8266-scheduler\examples\SchedulerBlink\SchedulerBlink.ino
Using board 'nodemcuv2' from platform in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hannah\AppData\Local\Temp\arduino_build_116793/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\variants\nodemcu" "C:\Users\hannah\AppData\Local\Temp\arduino_build_116793\sketch\SchedulerBlink.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\ [... cut ...] cu" "C:\Users\hannah\AppData\Local\Temp\arduino_build_116793\sketch\SchedulerBlink.ino.cpp" -o "C:\Users\hannah\AppData\Local\Temp\arduino_build_116793\preproc\ctags_target_for_gcc_minus_e.cpp"
D:\esp8266-scheduler-master\esp8266-scheduler\examples\SchedulerBlink\SchedulerBlink.ino:24:23: fatal error: Scheduler.h: No such file or directory

 #include <Scheduler.h>

                       ^

compilation terminated.

exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

closed 2018-02-13 by anmaped

shiftleftplusone, 2018-02-14:
why doesn't it work with 1.8.5 ?
1.6.8 is outdated, and for current esp8266 projects including different other libs  I also always use 1.8.5