anmaped/esp8266-scheduler #3: please rename this scheduler e.g. to esp8266Scheduler

shiftleftplusone, 2017-09-06:
just observed that this anmaped/esp8266-scheduler has the same lib name "Scheduler" ( #include <Scheduler.h> ) as the cmaglie (and the origonal mpatil) version which is already installed on my system and would overwrite them immediately - so it's no choice unless there would be a way to rename it before (e.g., #include <esp8266Scheduler.h> )

anmaped, 2017-10-06:
Could you elaborate more on the issue? Your environment etc..

shiftleftplusone, 2017-10-18:
I am working with several boards, e.g. also with the Arduino Due and the Due Scheduler by c. maglie, and now also with the esp8266. Both libs have to be in the libraries folder (Sketchbook or portable) and so they must have different names.

shiftleftplusone, 2018-01-13:
I still can't run the esp8266-Scheduler as I already have installed the Scheduler form C. Maglie.
Trying to use esp8266-Scheduler I get the error

> 
WARNUNG: Bibliothek Scheduler behauptet auf (sam, samd) Architektur(en) ausgeführt werden zu können und ist möglicherweise inkompatibel mit Ihrem derzeitigen Board, welches auf (esp8266) Architektur(en) ausgeführt wird.

So your Scheduler lib does not work when the Due scheduler has also been installed.

Most probably this issue happens because both libs are included by 
`#include <Scheduler.h>`
 So again. please rename your entire Scheduler lib files by esp8266-Scheduler, so that it can be included by
`#include <esp8266-Scheduler.h> // <===== !!! `
so that it will not falsely interfere with C. Maglie's  Scheduler by identical, ambiguous
`#include <Scheduler.h>`

per1234, 2018-02-10:
@shiftleftplusone please try again with the latest version of this library:
https://github.com/anmaped/esp8266-scheduler/archive/master.zip

All recent versions of the Arduino IDE do architecture based include priority inversion. This means the library with library.properties `architectures` value matching the architecture of the board you're compiling for gets include priority. The `architectures` value of the official Arduino Scheduler library is:
```
architectures=sam,samd
```
This library's `architectures value is:
https://github.com/anmaped/esp8266-scheduler/blob/498112d0ca3d16b85838c1ab9a2a34174a52155c/library.properties#L9

Therefore this library will have include priority when compiling for ESP8266 and the issue you report should not occur. My suspicion is that you installed an version of the library that's over a year old when it didn't have `esp8266` in the `architectures` list.

shiftleftplusone, 2018-02-13:
still does not work.
folder  esp8266-scheduler-master renamed to esp8266-scheduler and copied into the potable\libraries folder
Sketch Scheduler.blink copied to d:\ for current use.

Code: Scheduler.blink
```
/**
 * @file SchedulerBlink.ino
 * @version 1.0
 *
 * @section License
 * Copyright (C) 2015-2016, Mikael Patel
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * @section Description
 * This Arduino sketch uses the Scheduler library to blink four
 * LEDs with different speed. Each LED is controlled by a separate
 * task.
 */

#include <Scheduler.h>

const int led = 13;

void setup()
{
  pinMode(led, OUTPUT);

  Scheduler.start(setup1, loop1);
  Scheduler.start(setup2, loop2);
  Scheduler.start(setup3, loop3);
}

void loop()
{
  blink(led, 1000);
}

const int led1 = 10;

void setup1()
{
  pinMode(led1, OUTPUT);
}

void loop1()
{
  blink(led1, 250);
}

const int led2 = 11;

void setup2()
{
  pinMode(led2, OUTPUT);
}

void loop2()
{
  blink(led2, 500);
}
[... 16 lines cut ...]
  delay(ms);
  digitalWrite(pin, LOW);
  delay(ms);
}
```
errors:
Arduino: 1.8.5 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Program Files (x86)\Arduino\portable\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Program Files (x86)\Arduino\portable\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\portable\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path C:\Users\hannah\AppData\Local\Temp\arduino_build_116793 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\esptool\0.4.9 -verbose D:\esp8266-scheduler-master\esp8266-scheduler\examples\SchedulerBlink\SchedulerBlink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Program Files (x86)\Arduino\portable\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Program Files (x86)\Arduino\portable\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\portable\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path C:\Users\hannah\AppData\Local\Temp\arduino_build_116793 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\esptool\0.4.9 -verbose D:\esp8266-scheduler-master\esp8266-scheduler\examples\SchedulerBlink\SchedulerBlink.ino
Using board 'nodemcuv2' from platform in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hannah\AppData\Local\Temp\arduino_build_116793/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\variants\nodemcu" "C:\Users\hannah\AppData\Local\Temp\arduino_build_116793\sketch\SchedulerBlink.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\ [... cut ...] cu" "C:\Users\hannah\AppData\Local\Temp\arduino_build_116793\sketch\SchedulerBlink.ino.cpp" -o "C:\Users\hannah\AppData\Local\Temp\arduino_build_116793\preproc\ctags_target_for_gcc_minus_e.cpp"
D:\esp8266-scheduler-master\esp8266-scheduler\examples\SchedulerBlink\SchedulerBlink.ino:24:23: fatal error: Scheduler.h: No such file or directory

 #include <Scheduler.h>

                       ^

compilation terminated.

exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

closed 2018-02-13 by anmaped

shiftleftplusone, 2018-02-14:
why is the issue closed?
the blink sketch coming with this lib esp8266-scheduler forked from mikaelpatel/Arduino-Scheduler cannot be compiled.

per1234, 2018-02-14:
You've been working with Arduino for at least three years now. How is it possible that you don't understand the meaning of `fatal error: Scheduler.h: No such file or directory`?

shiftleftplusone, 2018-02-14:
the example blink source is not from me but just is as-it-is.
Since 3 years I am just a Arduino hobby user, no expert or professional.
I installed the library  esp8266-scheduler as described into the libraries folder.
Eventually I only tried to compile a given example file, which is expected to work, but it does not work.
Why does the given  lib example not work then?