anmaped/esp8266-scheduler #5: SchedulerBlink does not compile for esp8266

shiftleftplusone, 2018-01-13:
SchedulerBlink does not compile for esp8266. Lotc of errors.

note:
the esp8266-scheduler  is the only Scheduler lib currently, all different ones (e.g. by c.maglie) have been removed temporarily.

Arduino 1.8.5

> 
> C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Program Files (x86)\Arduino\portable\packages -hardware D:\Arduino\patch_libs\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Program Files (x86)\Arduino\portable\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\patch_libs\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path C:\Users\hannah\AppData\Local\Temp\arduino_build_758314 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\esptool\0.4.9 -verbose D:\SchedulerBlink\SchedulerBlink.ino
> C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Program Files (x86)\Arduino\portable\packages -hardware D:\Arduino\patch_libs\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Program Files (x86)\Arduino\portable\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\patch_libs\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path C:\Users\hannah\AppData\Local\Temp\arduino_build_758314 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\esptool\0.4.9 -verbose D:\SchedulerBlink\SchedulerBlink.ino
> Using board 'nodemcuv2' from platform in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0
> Using core 'esp8266' from platform in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0
> Detecting libraries used...
> "C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hannah\AppData\Local\Temp\arduino_build_758314/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\variants\nodemcu" "C:\Users\hannah\AppData\Local\Temp\arduino_build_758314\sketch\SchedulerBlink.ino.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hannah\AppData\Local\Temp\arduino_build_758314/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\variants\nodemcu" "-ID:\Arduino\patch_libs\Arduino\libraries\esp8266-scheduler-master" "C:\Users\hannah\AppData\Local\Temp\arduino_build_758314\sketch\SchedulerBlink.ino.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hannah\AppData\Local\Temp\arduino_build_758314/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\variants\nodemcu" "-ID:\Arduino\patch_libs\Arduino\libraries\esp8266-scheduler-master" "D:\Arduino\patch_libs\Arduino\libraries\esp8266-scheduler-master\Scheduler.cpp" -o "nul"
> Generating function prototypes...
> "C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hannah\AppData\Local\Temp\arduino_build_758314/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\variants\nodemcu" "-ID:\Arduino\patch_libs\Arduino\libraries\esp8266-scheduler-master" "C:\Users\hannah\AppData\Local\Temp\arduino_build_758314\sketch\SchedulerBlink.ino.cpp" -o "C:\Users\hannah\AppData\Local\Temp\arduino_build_758314\preproc\ctags_target_for_gcc_minus_e.cpp"
> "C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\hannah\AppData\Local\Temp\arduino_build_758314\preproc\ctags_target_for_gcc_minus_e.cpp"
> Sketch wird kompiliert...
> "C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hannah\AppData\Local\Temp\arduino_build_758314/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program [... cut ...]
> C:\Users\hannah\AppData\Local\Temp\arduino_build_758314\libraries\esp8266-scheduler-master\Scheduler.cpp.o: In function `SchedulerClass::init(void (*)(), void (*)(), unsigned char const*)':
> 
> Scheduler.cpp:(.text._ZN14SchedulerClass4initEPFvvES1_PKh+0x1f): undefined reference to `setjmp'
> 
> collect2.exe: error: ld returned 1 exit status
> 
> Bibliothek esp8266-scheduler-master in Version 1.2.1 im Ordner: D:\Arduino\patch_libs\Arduino\libraries\esp8266-scheduler-master  wird verwendet
> exit status 1
> Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).
>

per1234, 2018-02-06:
Duplicate of https://github.com/anmaped/esp8266-scheduler/issues/1

After years of submitting issues @shiftleftplusone (AKA VogonJeltz) still hasn't figured out how to use an issue tracker.

shiftleftplusone, 2018-02-07:
better fix the issues and abandon those rediculous accusations and thumb-down-ratings.

anmaped, 2018-02-07:
Duplicated.

closed 2018-02-07 by anmaped

shiftleftplusone, 2018-02-13:
still does not work.
folder  esp8266-scheduler-master.zip downloaded from
https://github.com/anmaped/esp8266-scheduler/archive/master.zip
extracted and  renamed to esp8266-scheduler and copied into the potable\libraries folder
Sketch Scheduler.blink copied to d:\ for current use.

Code: Scheduler.blink
```
/**
 * @file SchedulerBlink.ino
 * @version 1.0
 *
 * @section License
 * Copyright (C) 2015-2016, Mikael Patel
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * @section Description
 * This Arduino sketch uses the Scheduler library to blink four
 * LEDs with different speed. Each LED is controlled by a separate
 * task.
 */

#include <Scheduler.h>

const int led = 13;

void setup()
{
  pinMode(led, OUTPUT);

  Scheduler.start(setup1, loop1);
  Scheduler.start(setup2, loop2);
  Scheduler.start(setup3, loop3);
}

void loop()
{
  blink(led, 1000);
}

const int led1 = 10;

void setup1()
{
  pinMode(led1, OUTPUT);
}

void loop1()
{
  blink(led1, 250);
}

const int led2 = 11;

void setup2()
{
  pinMode(led2, OUTPUT);
}

void loop2()
{
[... 18 lines cut ...]
  delay(ms);
  digitalWrite(pin, LOW);
  delay(ms);
}
```
errors:
Arduino: 1.8.5 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Program Files (x86)\Arduino\portable\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Program Files (x86)\Arduino\portable\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\portable\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path C:\Users\hannah\AppData\Local\Temp\arduino_build_116793 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\esptool\0.4.9 -verbose D:\esp8266-scheduler-master\esp8266-scheduler\examples\SchedulerBlink\SchedulerBlink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Program Files (x86)\Arduino\portable\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Program Files (x86)\Arduino\portable\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\portable\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path C:\Users\hannah\AppData\Local\Temp\arduino_build_116793 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\esptool\0.4.9 -verbose D:\esp8266-scheduler-master\esp8266-scheduler\examples\SchedulerBlink\SchedulerBlink.ino
Using board 'nodemcuv2' from platform in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hannah\AppData\Local\Temp\arduino_build_116793/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\variants\nodemcu" "C:\Users\hannah\AppData\Local\Temp\arduino_build_116793\sketch\SchedulerBlink.ino.cpp" -o "nul"
[... cut ...] cu" "C:\Users\hannah\AppData\Local\Temp\arduino_build_116793\sketch\SchedulerBlink.ino.cpp" -o "C:\Users\hannah\AppData\Local\Temp\arduino_build_116793\preproc\ctags_target_for_gcc_minus_e.cpp"
D:\esp8266-scheduler-master\esp8266-scheduler\examples\SchedulerBlink\SchedulerBlink.ino:24:23: fatal error: Scheduler.h: No such file or directory

 #include <Scheduler.h>

                       ^

compilation terminated.

exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).