anmaped/esp8266-scheduler #6: Unable to compile (Scheduler.h)

kkirjala, 2018-02-21:
Using Arduino IDE 1.8.5 and ESP8266 (LinkSprite LinkNode D1, to be precise) I get following compile error:

/home/kalle/Arduino/libraries/esp8266-scheduler/Scheduler.h:138:19: error: expected ';' before 'sched'
 #define Scheduler sched()

I'm using the current (as of Feb 20th 2018) version of esp8266-scheduler with this modified esp8266 runtime: https://github.com/srmq/Arduino/tree/scheduler_branch

kkirjala, 2018-02-21:
Sorry, I believe this was made in error.

closed 2018-02-21 by kkirjala

anmaped, 2018-02-21:
@kkirjala  Is it working?

kkirjala, 2018-02-21:
Yes. Seems so. 

I don't know what the issue was. I fixed it by copy-pasting code (as seen on Github when browsing the source files on web site) on top of that. I hadn't modified the file previously. I'll reopen in case it persists.

anmaped, 2018-02-21:
@kkirjala  Probably, it is related to the line endings.